Nishida-Lab/rosbook_pkgs #20: 変数等の命名則について

yamacir-kit, 2018-07-14:
C++のサンプルコードですが，変数名はかなり冗長になってしまったとしても，略称を用いない名前を使ったほうが良いと考えてるのですが，いかがでしょうか．
（変数名自体の命名則についてはコーディングスタイルガイドでも言及されていません）

多くの場合，ノードハンドルは `ros::NodeHandle nh` とすることが慣習となっており，ROS開発者にとっては馴染み深いものですが，
この本が対象とする読者はROSの初心者ということなので，例えば先の例で言えば `ros::NodeHandle node_handle` としておいたほうが，
読者が「nhってなんだっけ」とソースコードを探したり思い出したりする負担を減らせると思います．

その際は，注釈として
「通常は入力を楽にするため `nh` などの略称を用いるが，本書ではわかりやすさを重視して変数名をつけている」
という旨の文を併記すれば親切だと思います．

MoriKen254, 2018-07-14:
改善のご提案ありがとうございます！

ご指摘の件、すごーくよくわかります。（と言っている小生の作成したコードでも NodeHandle を nと宣言してしまっております汗。当時はそこまで意識しきれていなかったみたいです汗。）

私も個人でコーディングするとき、ご指摘の命名規則を意識している身分ですので、抱かれているモヤモヤ感もとってもよくわかります。

一つ悩みがあるとすると、命名規則に踏み込むと本文への影響が大きくなるということです。

変数名を差し替え→本文中のソースコード差し替え→本文で変数の説明している部分を差し替え

という具合の修正が、サンプルが存在する全ての章で必要になる（C++、Python問わず両方）ということです。

せめて、修正範囲をカプセル化するくらいの修正に留める必要があります。全編に渡り省略型の命名が慣習的に成り立ってしまっていることは許容しなければならないかもしれません…。

例えば、NodeHandle型の変数名がnだったりnhだったりするのを、せめてnhに統一する等、このあたりに留めることは、ひとまず落とし所かと思われます。

そのようなことを考慮した上で、Stepを後述の3段階に分けてpushしてもらうのはいかがでしょう（優先度の高いものから順に進めていく構成になっています）。編集者（西田先生）としても、QCDのバランスを考慮して、どこまでできるか判断した上で選択できると思います。

少し大変なお願いとなっていることは承知で申し訳ないのですが、ご検討頂ければと思います。

## Step1. フォーマットの整形のみ

変数名等は据え置きで、とにかくスタイルフォーマットのみの整形にとどめた修正。

基本として、本文への影響はソースコードの差し替えのみ。変更が生じる場合はその箇所のリストを提出。

## Step2. 変数名のゆらぎの修正

省略的な変数名は据え置きのまま、もしStep1の修正完了後に、統一できていない変数名がある場合のみの箇所を修正。

本文への影響は軽微。ただし、変更した場合に、本文何ページのどの箇所を差し替える必要があるかを示した全リストの提出（エクセルが望ましい）、あるいは本文のpdfの全ての該当箇所に漏れがないようにコメントを入れて先生に提出。

## Step3. 命名規則の完全準拠

全編のソースコード(C++, Python両者)全ての変数名、クラス名、名前空間名等につき、説明的な命名規則に完全準拠させたもの。

こちらも同様に、変更箇所の全リストをもれなく提出。

yamacir-kit, 2018-07-14:
@MoriKen254 

本文の修正の方を失念していました．
すみません．

上記の方針で進めていきます．

ひとつのステップを終える毎にmasterへプルリクを送るようにしようと思います．
その際は先生に送るexcelリストと同等の内容をプルリク本文に加える予定です．

closed 2018-07-15 by yamacir-kit