Nishida-Lab/rosbook_pkgs #21: fourth_robot.urdf.xacro が読めない

MoriKen254, 2018-07-15:
取り急ぎ。

## 推定原因
確かxacroの文法がKineticで若干変更になったはず。Indigoに合わせて記述したfource_robotがそれに適合できていないと思われる。

## コマンド

```bash
roslaunch fourth_robot_description fourth_robot_description.launch 
```

## エラーメッセージ

```bash
... logging to /home/masa/.ros/log/22dde34a-87ce-11e8-8779-000c29a746a8/roslaunch-ubuntu-3808.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /home/masa/rosbook_ws/src/rosbook_pkgs/chapter6/fourth_robot_description/robots/fourth_robot.urdf.xacro

unknown macro name: xacro:wheel_gazebo_control_v0 None None
when processing file: /home/masa/rosbook_ws/src/rosbook_pkgs/chapter6/fourth_robot_description/robots/fourth_robot.urdf.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/masa/rosbook_ws/src/rosbook_pkgs/chapter6/fourth_robot_description/robots/fourth_robot.urdf.xacro'] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py '$(arg model)'" name="robot_description"/>
The traceback for the exception was written to the log file
```

yamacir-kit, 2018-07-15:
それはそこに書かれている通りのコマンドを実行すれば解決します
`find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'`

MoriKen254, 2018-07-15:
#23 で解決。ご協力ありがとうございました！

closed 2018-07-15 by MoriKen254