Nishida-Lab/rosbook_pkgs #34: 書籍としての改善案・要望等

MoriKen254, 2018-09-19:
発売前からこんなことを書くのかと思われるかもしれませんが、2版以降改善できるところ、あるいは読者からの要望等があればここに記しましょう。

一時的には、本リポジトリで正誤表的にアナウンスすることはあると思います。

- [ ] 共著者・読者からの追加要望
  - 残件があり、かつもし読者のご意見も頂けたなら、落ち着いた段階でまとめて腰を据えて反映させましょう。
- [ ] Kindle 化
  - https://github.com/Nishida-Lab/rosbook_pkgs/issues/33 で述べたとおり。
- [ ] MoveIt の章の追加
  - RISを提案している西田研発の本だし、書籍のキャッチにも「産ロボ」とアピールされているのに、本書にMoveit!のチュートリアル解説が無いのは若干気にはなっていました（スキルのある人間は中にいるのに、もったいない！的なラフな感じですが）。
  - とは言え、まずは出版が先決でありましたし、ページ数の制約等を配慮し、優先度の高い納品済み原稿の修正を重視しました。
  - 既存のmotoman_projectのリポジトリのうちシンプルなモデルを流用して、Kinectを利用した物体認識とピック&プレースの簡単なデモをGazeboで行うチュートリアルとか、かなり需要はあると思います（やっぱりもったいなかったかな！笑）。

色々やると価格に影響するかもしれませんが、まずは市場の反応を見て導入を検討したく。

よろしくお願い致します。

RyodoTanaka, 2018-09-20:
@MoriKen254 
おっしゃるとおり，Moveit!に関してのデモはあると良いと僕も思います．
Moveit!のmotoman_projectに関しては，かなり西田研に特化しすぎているというのと，ごちゃごちゃでわかりづらいという理由から，以前ご紹介したとおり [ntl-ros-pkgs](https://github.com/ntl-ros-pkg)なるページにてまとめ直したものを公開しています．
Pick&Placeを行うデモであればあとは
#### 必ず必要になるもの
- 物体認識パッケージの追加（Moveit! 純正のを使うのも良いかも）
- ハンドのGraspingに関するプラグインの追加（D-Handそのままの開閉では持てないので，"吸い付け"プラグイン https://github.com/JenniferBuehler/gazebo-pkgs/wiki/The-Gazebo-grasp-fix-plugin が必要...?）
- Pick&Placeを行うデモコード

#### あったらいいね！というもの
- 状態機械による状態遷移によるPick&Placeモード（smaschを使う？ROSConJPのLT発表であった[event_manager](https://github.com/OUXT-Polaris/event_manager)も良いかも）

という感じかと思ってます．
よろしくお願いします．

MoriKen254, 2018-09-20:
ありがとう！

https://github.com/ntl-ros-pkg はちょっと拝借してみますわ笑。ちょうどシンプルなの使いたかったんだわ。

あと、DHand使うとなると結構敷居が高くなるから、もっとシンプルなチャック型のハンドのサンプルがあればいいかなって。それなら吸い付け関連いらなくなるし、Graspingのプラグインの解説に注力できるかなーって。

あと、個人的には、プランナーのMoveItプラグイン作成が気になるなぁ。ROS実証を検討する企業では需要あると思う。あれ結構大変かな？

RyodoTanaka, 2018-09-21:
@MoriKen254 
> もっとシンプルなチャック型のハンドのサンプルがあればいいかなって。それなら吸い付け関連いらなくなるし、Graspingのプラグインの解説に注力できるかなーって。

僕もこれに賛成です！
洋書のROS本だと，適当な並行チャックを作ってあったと思うんですけど，実在する並行チャックをモデルにするとより喜ばれるかなぁと思ってます．
西田研にあるチャックをモデルにして，実機とも合わせ込もうと思ったのですが，空気圧回路が結局出来上がってないので，試せてない状態です...orz
また，並行チャックだけでなく，単純に吸い付け型のハンドも需要ありそうだなと思うので，考慮に入れとくと良いかもしれない，とも思ってます．

> プランナーのMoveItプラグイン作成が気になるなぁ。ROS実証を検討する企業では需要あると思う。あれ結構大変かな？

ドキュメントはこれですかね？
https://moveit.ros.org/documentation/plugins/

OMPLにオリジナルのプランナを追加して，Moveit!に組み合せる方法なら経験があるのでできます．
ただ，Moveit!に限らずOMPLにも話が及ぶので，結構大変だと思います．

MoriKen254, 2018-09-21:
> 西田研にあるチャックをモデルにして，実機とも合わせ込もうと思ったのですが，空気圧回路が結局出来上がってないので，試せてない状態です...orz

Gazeboでシミュレートする際にも空気圧回路必要ですか？テケトーなros_controller割り当てて開閉だけしたいなと思ったとき、何らかの技術ネックがあるということですよね。

> また，並行チャックだけでなく，単純に吸い付け型のハンドも需要ありそうだなと思うので，考慮に入れとくと良いかもしれない，とも思ってます．

HSRがやってたやつですね。よっしゃ、そういうことなら両方やるぞ！そのパッションと自発性が一番大事！

> Moveit!に限らずOMPLにも話が及ぶので，結構大変だと思います．

あー、なるほど。そうなると一章で完結させるのは厳しそうですね。解説としては、Pick & Place まででにして、この辺については手を引きましょう。

***

自発的にブランチ切ってサンプル入れる分には問題ないと思います。Dropbox で自発的に原稿を追加するのも問題ないと思います。あとはそれらのリソースをしかるべきタイミングにしかるべき場所でリリースしましょう。必ずしも「書籍化しなければならない」ということではないはずなので。

一義的に大切なのは、西田研で培われたノウハウを、他のデベロッパと共有し、ROSの発展に貢献すること。その上で、二義的に著者のプレゼンスが向上されるのが望ましい。書籍化は、後者のインパクトが大きいという点で優位性があります。

私達デベロッパに好き勝手やらせて下さる踊り場を整備して下さった西田先生には、共著者としての最大級の感謝の念を忘れてはいけないと思います。

RyodoTanaka, 2018-11-15:
@MoriKen254 
西田先生よりメールがあったと思いますが，重版に伴って修正が可能です．
- `steer_drive_controller`の部分
- `smach_sample`のobject_spawnの部分

が今のところ修正したら良さそうと思いますが，いかがでしょうか？
他にもありましたらここに記します．
よろしくお願いします．

MoriKen254, 2018-11-16:
@RyodoTanaka 
そうですね。それは要望として出しましょう。

他に気になるものがあったような記憶があるので、一回読みなおしてみますね。