Nishida-Lab/rosbook_pkgs #37: 【10章】オブジェクトがテーブル下にspawnされる

MoriKen254, 2018-10-26:
題記、https://github.com/Nishida-Lab/rosbook_pkgs/issues/35 で言及。

おそらくですが、マシンスペックの影響？（当方では上にのっている）
table がspawnされる前に他のオブジェクトが自由落下でしたに行ってしまっている？

テーブルの上に乗せるなら、`chapter10/smach_sample/launch/spawn_object.launch`にて、下記スクリプトのz座標を高くすれば解決します。

```
    <node name="$(anon spawn2)" pkg="gazebo_ros" type="spawn_model" output="screen"
        args="-sdf -param beer_sdf -model beer_model -x 1.38 -y -0.1 -z 0.66" />
    <node name="$(anon spawn3)" pkg="gazebo_ros" type="spawn_model" output="screen"
        args="-sdf -param plastic_cup_sdf -model cup_model -x 1.38 -y 0.2 -z 0.66" />
```

ただ、あまり高くするとテーブルに載ったときにグラスが倒れます。当方の環境ではこの高さがテーブルに乗り、かつグラスが倒れないいい感じの値だったので、これにしました。

@AtsukiYokota さんの環境でうまくいく値に修正して報告してもらえますか？

AtsukiYokota, 2018-10-27:
当方の環境で試して見た結果テーブルよりも先にビール缶とコップがspawnしてしまい、テーブルの上に載らないようです。ビール缶やコップをテーブルのspawnに間に合うようにz座標を高くすると今度は高すぎてコップ等が倒れてしまいます。
根本的な解決にはなっていませんがテーブルとオブジェクト(ビール缶とコップ)を別のlaunchファイルに分けてそれぞれ手動で時間差でlaunchさせると確実にテーブルの上にオブジェクトが載ります。(当たり前ですが・・・)
テーブルとオブジェクトをspawnさせるlaunchファイルはそれぞれ以下のように記述しました。
```spawn_table.launch
<launch>
    <param name="table_sdf" textfile="$(find smach_sample)/sdf/table/model-1_4.sdf" />
    <node name="$(anon spawn1)" pkg="gazebo_ros" type="spawn_model" output="screen"
    args="-sdf -param table_sdf -model table_model -x 1.6 -z -0.4 -Y 1.570" />
</launch>
```
```spawn_object.launch
<launch>
    <param name="beer_sdf" textfile="$(find smach_sample)/sdf/beer/model-1_4.sdf" />
    <param name="plastic_cup_sdf" textfile="$(find smach_sample)/sdf/plastic_cup/model.sdf" />
    <node name="$(anon spawn2)" pkg="gazebo_ros" type="spawn_model" output="screen"
    args="-sdf -param beer_sdf -model beer_model -x 1.38 -y -0.1 -z 0.65" />
    <node name="$(anon spawn3)" pkg="gazebo_ros" type="spawn_model" output="screen"
    args="-sdf -param plastic_cup_sdf -model cup_model -x 1.38 -y 0.2 -z 0.65" />
</launch>
```

@MoriKen254 さんの`timed_roslaunch`を用いればもう少しスマートかもですが、どう対処するかはお任せしようと思います。

MoriKen254, 2018-10-28:
ありがとうございます。状況はよくわかりました。tableは構成要素が大きいから、spawnerの応答速度がマシンスペック次第では大きく遅延してしまう場合があるのだと推察されます。

では、次のような方針にするのはどうでしょう。

### 現在のlaunchは据え置き
ROS本の本文で全文が記述されているファイルなので、これを消すわけには行きません。
よって、据え置きします。

### その上で launch を分けたファイルも用意
@AtsukiYokota さんご指摘の分割型のファイルも用意します。その使用方法は、READMEに記載します。

### 更に、timed_loslaunchのファイルも用意
よくご存知でしたね笑。

これを使うと解決する問題って結構多いのは分かるのですが、教育用書籍という観点から、あまり外部パッケージを入れることで純粋な学習を妨げてしまわないか懸念している節がありました。

とはいえ、こういう状況ですし、「実用」という名前も入れていますし、リポジトリにwstoolを追加して入れるのもいいかな、と思っています。

そして、もう一つlaunchを追加し、READMEに記載。この方針なら、ROS本との整合性も保った上で、もう少し中身が分かる人向けの応用ユースケースの紹介もできる。

あるいは、upstreamのaptに追加したいかも。aptに追加するの結構大変な記憶があるのですが、やり方
分かる人是非教えて下さい。

自作pkgを過大評価するつもりはないのですが、timed_roslaunchは、その価値があると思います。

MoriKen254, 2018-10-28:
> upstreamのaptに追加したい

この辺ですよねぇ。

- Release
  - http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease

- PreRelease
  - http://wiki.ros.org/regression_tests#How_do_I_setup_my_system_to_run_a_prerelease.3F

怖いのでPreReleaseから着手しているのですが、すこし手こずりそうなので、launch分割＋READMEまでで一旦PR出します。

MoriKen254, 2018-11-01:
TORKのブログでリリース関連の情報共有がされています。やってみます。

https://opensource-robotics.tokyo.jp/?p=3272

MoriKen254, 2018-11-05:
自作パッケージをリリースするにあたって調べてみたメモです。情報があちゃこちゃにあって目がシパシパしていたのですが笑、ようやっと`Buildfarm`と`bloom`の仕組みが見えてきました汗。

特にまとめているわけではありませんが、備忘録として（すぐ忘れるので笑）、かつ情報共有として一旦簡潔に整理したものを上げておきます。

# ROSCon Presentation
- https://vimeo.com/187705230
- https://roscon.ros.org/2016/presentations/ROSCon2016%20Build%20Farm.pdf

# Debian Samples
## ros_controllers
- Wiki
  - http://wiki.ros.org/ros_controllers

- Source (sourcedev)
  - http://build.ros.org/view/Msrc_uB/job/Msrc_uB__ros_controllers__ubuntu_bionic__source/4/console#console-section-9

- Binary (binarydev)
  - http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__ros_controllers__ubuntu_bionic_amd64__binary/

- Devel
  - http://build.ros.org/view/Mdev/job/Mdev__ros_controllers__ubuntu_bionic_amd64/17/console

- Doc
  - http://build.ros.org/view/Mdoc/job/Mdoc__ros_controllers__ubuntu_bionic_amd64/15/console#console-section-10

- setting yaml file

>　ros_controllers:
　　doc:
　　　type: git
　　　url: https://github.com/ros-controls/ros_controllers.git
　　　version: indigo-devel
　　release:
　　　packages:
　　　- diff_drive_controller
　　　- effort_controllers
　　　- force_torque_sensor_controller
　　　- forward_command_controller
　　　- gripper_action_controller
　　　- imu_sensor_controller
　　　- joint_state_controller
　　　- joint_trajectory_controller
　　　- position_controllers
　　　- ros_controllers
　　　- rqt_joint_trajectory_controller
　　　- velocity_controllers
　　　tags:
　　　　release: release/indigo/{package}/{version}
　　　url: https://github.com/ros-gbp/ros_controllers-release.git
　　　version: 0.9.4-0
　　source:
　　　type: git
　　　url: https://github.com/ros-controls/ros_controllers.git
　　　version: indigo-devel
　　　status: developed

## steer_drive_ros
- Wiki
  - http://wiki.ros.org/steer_drive_ros?distro=indigo

- Devel
  - http://build.ros.org/view/Idev/job/Idev__steer_drive_ros__ubuntu_trusty_amd64/6/console

- Doc
  - http://build.ros.org/view/Idoc/job/Idoc__steer_drive_ros__ubuntu_trusty_amd64/3/console

- setting yaml file

>　steer_drive_ros:
　　doc:
　　　type: git
　　　url: https://github.com/CIR-KIT/steer_drive_ros.git
　　　version: indigo-devel
　　source:
　　　type: git
　　　url: https://github.com/CIR-KIT/steer_drive_ros.git
　　　version: indigo-devel
　　status: maintained

## timed_roslaunch
- Release
  - https://github.com/MoriKen254/timed_roslaunch-release
- Wiki
  - http://wiki.ros.org/timed_roslaunch

>　timed_roslaunch:
　　doc:
　　　type: git
　　　url: https://github.com/MoriKen254/timed_roslaunch.git
　　　version: kinetic-devel
　　release:
　　　tags:
　　　　release: release/kinetic/{package}/{version}
　　　url: https://github.com/MoriKen254/timed_roslaunch-release.git
　　　version: 0.1.1-3
　　source:
　　　test_pull_requests: true
　　　type: git
　　　url: https://github.com/MoriKen254/timed_roslaunch.git
　　　version: kinetic-devel
　　status: maintained

# ROS-I
- https://rosindustrial.org/developmentprocess/

# Bloom
## Home Page
- http://ros-infrastructure.github.io/bloom/
- https://bloom.readthedocs.io/en/0.5.10/

## ROSWiki
- http://wiki.ros.org/bloom
- http://wiki.ros.org/build.ros.org
- http://wiki.ros.org/buildfarm
- http://wiki.ros.org/buildfarm/Design

## Tutorial
- http://wiki.ros.org/ja/bloom/Tutorials/FirstTimeRelease

## GitHub
- https://github.com/ros-infrastructure/buildfarm_deployment

## REP 143
- http://www.ros.org/reps/rep-0143.html

# Jenkins
- http://megadreams14.com/?p=27
- https://cloudbees.techmatrix.jp/jenkins/

# Buildfarm sample
## apt list
- Top
  - http://packages.ros.org/ros/ubuntu/

- Source
  - http://packages.ros.org/ros/ubuntu/lists/ros-kinetic-xenial-source_xenial_main_Sources

- Binary
  - http://packages.ros.org/ros/ubuntu/lists/ros-kinetic-xenial-arm64_xenial_main_arm64_Packages

## diff_drive_controller
- Source
  - http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-diff-drive-controller/ 
ros-kinetic-diff-drive-controller_0.13.4.orig.tar.gz 

- Binary
  - http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-diff-drive-controller/ 
 ros-kinetic-diff-drive-controller_0.13.4-0xenial-20181005-135508-0800_amd64.deb

## Install
- Lists
  - http://wiki.ros.org/kinetic/Installation/Ubuntu

- Source
  - http://wiki.ros.org/DebianPackageSources

- Binary
  - http://wiki.ros.org/kinetic/Installation/Ubuntu

MoriKen254, 2018-11-08:
- 2018/11/08
  - `shadow-fix` に入る。
    - http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-kinetic-timed-roslaunch/

MoriKen254, 2018-11-08:
- 2018/11/08-2

[この辺](http://wiki.ros.org/ShadowRepository)参考にして`ros-shadow-fixed` 経由で`apt install`してみたのですが、`install`先がおかしくて参りました。コマンド自体は良さげなんですが、deb パッケージの中身が変な感じです。

```bash
$ sudo apt install ros-kinetic-timed-roslaunch
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
以下のパッケージが自動でインストールされましたが、もう必要とされていません:
  libllvm5.0
これを削除するには 'sudo apt autoremove' を利用してください。
以下のパッケージが新たにインストールされます:
  ros-kinetic-timed-roslaunch
アップグレード: 0 個、新規インストール: 1 個、削除: 0 個、保留: 141 個。
4,204 B のアーカイブを取得する必要があります。
この操作後に追加で 19.5 kB のディスク容量が消費されます。
取得:1 http://packages.ros.org/ros-shadow-fixed/ubuntu xenial/main amd64 ros-kinetic-timed-roslaunch amd64 0.1.1-3xenial-20181107-003648-0800 [4,204 B]
4,204 B を 0秒 で取得しました (8,988 B/s)               
以前に未選択のパッケージ ros-kinetic-timed-roslaunch を選択しています。
(データベースを読み込んでいます ... 現在 292355 個のファイルとディレクトリがインストールされています。)
.../ros-kinetic-timed-roslaunch_0.1.1-3xenial-20181107-003648-0800_amd64.deb を展開する準備をしています ...
ros-kinetic-timed-roslaunch (0.1.1-3xenial-20181107-003648-0800) を展開しています...
ros-kinetic-timed-roslaunch (0.1.1-3xenial-20181107-003648-0800) を設定しています ...
```

詳細を`ROS-ANSWERS`で質問中。承認待ちらしく、まだopenになってないかもしれません。助けてほしいです涙。。。
- https://answers.ros.org/question/307927/my-package-added-to-shadow-fixed-repository-with-bloom-is-installed-into-a-wrong-path/

MoriKen254, 2018-11-08:
`ROS-ANSWERS`で回答頂き、とりあえず仕切り直ししました。素早くてとてもありがたいです。

- 2018/11/09
  - release時に反映漏れコミットがあったのでバージョンアップとして再リリース

MoriKen254, 2018-11-08:
- 2018/11/09-2
  - Merged https://github.com/ros/rosdistro/pull/19427#event-1955595901

MoriKen254, 2018-11-09:
- 2018/11/09-3
    - `ros-shadow-fixed`にて`sudo apt install ros-kinetic-timed-roslaunch`のインストール先の修正を確認。

- 2018/11/10
    - しかし、ディレクトリ構造でまだ気になった点があったので、`gazebo_ros`のリポジトリを参考にしながらInstallの設定を見直し。0.1.3として再リリース。
    - https://github.com/ros/rosdistro/pull/19431

MoriKen254, 2018-11-10:
- 2018/11/10-2
  - `ros-shadow-fixed`にて`sudo apt install ros-kinetic-timed-roslaunch`を実施。所望の挙動を確認。
  - `Buildfarm` CI にてテスト成功確認。`ros-shadow-fixed`の段階でやってくれる。毎時45分のpolling。
    - http://build.ros.org/job/Kdev__timed_roslaunch__ubuntu_xenial_amd64/

MoriKen254, 2018-11-13:
- 2018/11/13
  - `public` に反映される http://packages.ros.org/ros/ubuntu/pool/main/r/
    - 前回更新が2018/10/30だったので、2週間でUpdateした模様。
    - `ros-shadow-fixed` だと平日火曜日～金曜日くらいは1～2日に1回ペースで反映される模様。
  - `sudo apt install ros-kinetic-timed-roslaunch` 成功。
    - とりあえず普通に動くっぽい。
    - サンプルスクリプトの実行権限が外されてしまっている。コピー先を見直して再リリース予定。

MoriKen254, 2018-11-13:
- 2018/11/14
  - テーブル spawn → 1秒後 ビールとカップ spawn OK。PR出す。

MoriKen254, 2018-11-16:
おかげさまで本Issueの目的が達成されました @ https://github.com/Nishida-Lab/rosbook_pkgs/pull/43

他distribution等への水平展開や、他パッケージの依存関係解決は、そちらのリポジトリのIssueに上げ、本件はCloseします。

ご協力いただき、ありがとうございました。