Nishida-Lab/rosbook_pkgs #39: 【10章】コップが浮いている

MoriKen254, 2018-10-27:
https://github.com/Nishida-Lab/rosbook_pkgs/pull/36 で言及。

> R2からみて左側の黒いコップが浮かんでるように見えます....

↓参考動画
https://goodroad.work/owncloud/index.php/s/kgD2tTVqMDFYEzX

MoriKen254, 2018-11-15:
https://github.com/Nishida-Lab/rosbook_pkgs/issues/37 のコードでspawnしたコップを見たところ、私の環境では浮いているようには見えなかったのですが、他の環境ではいかがでしょう？

RyodoTanaka, 2018-11-15:
@MoriKen254 
#43 のPR時に確認したのですが，確かにコップは浮いてない状態になっています．
下記がその時の動画です．
https://goodroad.work/owncloud/index.php/s/dR1naNds5X6J6u8
今考えると，`timed_roslaunch`を使用することにより，正しくモデルがロードされたため，きちんとスポーンできてるのではないかと思います．
浮いていた時のものは，そのときたまたま起きてしまったものかもしれません．
いずれにせよ，現在は浮くことなくスポーンしてます．

@NishidaLab
他の環境ではいかがでしょうか？

AtsukiYokota, 2018-11-16:
@MoriKen254 @RyodoTanaka 
私ともう一人の環境で試してみたところコップのspawnに関しては問題なくできました。
ただし、`timed_roslaunch`のrosdepによるインストール時に以下のようなエラーが出てインストールができない問題があります。

```
$ cd <catkin_ws>
$ rosdep install -iry --from-paths src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
smach_sample: Cannot locate rosdep definition for [timed_roslaunch]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

そのため今は手動でaptしてきて実行している状況です。

RyodoTanaka, 2018-11-16:
@AtsukiYokota 
それは，
```bash
$ rosdep update
```

した後にもう一度実行すれば解決します．

AtsukiYokota, 2018-11-16:
@RyodoTanaka 
ありがとうございます、rosdepの問題も解決しました。

MoriKen254, 2018-11-16:
みなさん、ご協力いただきまして、本当にありがとう。

正常動作を複数の環境で確認できたので、本件はクローズしますね。

closed 2018-11-16 by MoriKen254

RyodoTanaka, 2018-11-16:
@MoriKen254 @AtsukiYokota @西田研究室の皆様
本件解決したので閉じます．