Nishida-Lab/rosbook_pkgs #41: gitで落としたパッケージのビルドが通りません

Kenjiro-Minamikawa, 2018-11-06:
gitから落としたファイルの一部のビルドが通りません。

> git clone https://github.com/Nishida-Lab/rosbook_pkgs.git
Cloning into 'rosbook_pkgs'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (59/59), done.
remote: Total 1382 (delta 40), reused 64 (delta 25), pack-reused 1292
Receiving objects: 100% (1382/1382), 53.22 MiB | 1.79 MiB/s, done.
Resolving deltas: 100% (685/685), done.
Checking connectivity... done.

> catkin_make
Base path: /home/kenjiro/catkin_ws
Source space: /home/kenjiro/catkin_ws/src
Build space: /home/kenjiro/catkin_ws/build
Devel space: /home/kenjiro/catkin_ws/devel
Install space: /home/kenjiro/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/kenjiro/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/kenjiro/catkin_ws/build"
####
[  3%] Built target rosbag_example_publisher
[  7%] Built target rqtplot_example
[ 12%] Building CXX object rosbook_pkgs/chapter10/smach_sample/CMakeFiles/simple_gripper.dir/src/simple_gripper.cpp.o
[ 12%] Built target rosbag_example_subscriber
[ 16%] Built target opencv_ros
[ 27%] Built target pluginlib_arrayutil
[ 30%] Built target test_get_input
[ 34%] Built target test_edge_detection
[ 38%] Built target test_opencv
[ 41%] Built target test_feature_detection_matching
[ 45%] Built target test_feature_detection
[ 49%] Built target test_svm
[ 52%] Built target cloud_creator
[ 56%] Built target cloud_loader
[ 60%] Built target cloud_planar_segmenter
[ 63%] Built target tf2_static_broadcaster
[ 67%] Built target cloud_filter
[ 70%] Built target cloud_clusterer
[ 74%] Built target tf_listener
[ 78%] Built target cloud_downsampler
[ 81%] Built target cloud_matcher
[ 85%] Built target tf2_broadcaster
[ 89%] Built target tf2_listener
[ 92%] Built target tf_broadcaster
[ 98%] Built target client_node
/home/kenjiro/catkin_ws/src/rosbook_pkgs/chapter10/smach_sample/src/simple_gripper.cpp:2:58: fatal error: pr2_controllers_msgs/Pr2GripperCommandAction.h: No such file or directory
compilation terminated.
rosbook_pkgs/chapter10/smach_sample/CMakeFiles/simple_gripper.dir/build.make:62: recipe for target 'rosbook_pkgs/chapter10/smach_sample/CMakeFiles/simple_gripper.dir/src/simple_gripper.cpp.o' failed
make[2]: *** [rosbook_pkgs/chapter10/smach_sample/CMakeFiles/simple_gripper.dir/src/simple_gripper.cpp.o] Error 1
CMakeFiles/Makefile2:4824: recipe for target 'rosbook_pkgs/chapter10/smach_sample/CMakeFiles/simple_gripper.dir/all' failed
make[1]: *** [rosbook_pkgs/chapter10/smach_sample/CMakeFiles/simple_gripper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

どうすれば良いのかわかりません。
教えていただけると嬉しいです。

MoriKen254, 2018-11-06:
@Kenjiro-Minamikawa 様

お問い合わせありがとうございます。

今回のエラーは、依存パッケージである`pr2_controllers_msgs`がインストールされていないことが原因です。

書籍p.45 Gitの導入とサンプルプログラムの取得をご参考の上、`catkin_make`の前に下記コマンドをお試し頂けますか。

```bash
cd ~/<catkin_ws>
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -r -y
```

Kenjiro-Minamikawa, 2018-11-07:
治りました！
ありがとうございました！

MoriKen254, 2018-11-07:
解決したようで何よりです。issueを閉じます。

closed 2018-11-07 by MoriKen254