Nishida-Lab/rosbook_pkgs #48: 9章 9.2.1について

fumimaker, 2018-12-28:
本を使ってROSの勉強をさせていただいているものです．サンプルを動かしている中で動かない事象に遭遇したため，質問させて頂きたいです．
9.2.1の手動走行各種ノードの起動というところのjoy_controlを起動するところ
```
roslaunch joy_control joy_control.launch
```
で以下のようなエラーが出てしまいます．
```
Invalid <arg> tag: teleop_twist_joy
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/fumi/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share. 

Arg xml is <arg default="$(find teleop_twist_joy)/config/$(arg joy_config).config.yaml" name="config_filepath"/>
```
5章のJoyPadを使ったTurtleBotの操作はうまくいきましたが，ここでは何が問題なのでしょうか？

RyodoTanaka, 2018-12-28:
@fumimaker 

お問い合わせありがとうございます．
下記，お答えします．

## 原因
`teleop_twist_joy`パッケージが見当たらない事が原因です．
エラーメッセージ内の
```bash
ROS path .........
````
と続く部分のディレクトリを探索したが，`teleop_twist_joy`パッケージが見当たらなかった
というのが原因です．

## 対処法
ROSでは，各パッケージに記述されている`package.xml`を参照して，自動でインストールするツール（`rosdep`）があります．
今回も，依存パッケージがインストールされていなかったことが原因と考えられますので，これを使って対処します．
#### 1. `rosdep`のアップデート
```bash
$ rosdep update
```
#### 2. `<catkin_ws>`へ移動
```bash
$ cd <catkin_ws>
```

#### 3. 依存パッケージのインストール
```bash
$ rosdep install -i -y -r --from-paths src --rosdistro kinetic
```

ここまで完了後，再び
```bash
$ roslaunch joy_control joy_control.launch
```
を実行してみてください！

以上，よろしくおねがいします．

RyodoTanaka, 2018-12-28:
@fumimaker
先程の問題が解決したようで良かったです。
さて、お問い合わせ頂いた内容ですが、実はつい先日、全く同じIssueがありました。
#47 をご覧下さい！

fumimaker, 2018-12-28:
@RyodoTanaka 様
迅速なご回答ありがとうございます．
ちょうど今Issueを見直していたところ#47 を発見し，試していました．

結果としてはボタン8を押しながらアナログスティックを倒すことで問題なく動作するようになりました．joint-state-controllerとdiff-drive-controllerが足りていないのが原因のようでした．
ありがとうございました！

スレッドはクローズさせて頂きます．

closed 2018-12-28 by fumimaker