Nishida-Lab/rosbook_pkgs #52: USBカメラが使えない

daisuke0203, 2019-06-22:
第5章のp58にある
$ rosrun usb_cam usb_cam node
を実行すると以下のようなエラーが表示されます。
VIDIOC_STREAMON　error 5, Input/output error

USBカメラはLogicoolのC270です。

MoriKen254, 2019-06-22:
issue ありがとうございます。当方に同一のハードウェアがないため推測になりますが、回答致します。

rosrun で usb_cam ノードを引数無しで実行した場合、デフォルトのパラメータが適用された上で実行されると思われますが、そのデフォルト値がお使いになっているカメラの設定値と一致していないことによりエラーが生じている可能性が考えられます。

例えば、本書サンプル内の下記 launch ファイル内における、画像サイズやフレームレート等が、現在ご使用されている USB カメラの設定と矛盾していないかを、ご確認頂くことは可能でしょうか。

https://github.com/Nishida-Lab/rosbook_pkgs/blob/master/chapter5/launch/usb_cam.launch

エラーメッセージの情報量から、推定原因をベースとした議論となっているため的はずれな発言でしたらすみません。まずはご検証の方を宜しくおねがいします。

daisuke0203, 2019-06-25:
迅速な対応ありがとうございます。
以下、$ roslaunch chapter5 usb_cam.launchを実行した結果です。
daiuske@daiuske-VirtualBox:~$ roslaunch chapter5 usb_cam.launch
... logging to /home/daiuske/.ros/log/23a6245e-96e6-11e9-8d4a-080027134c6c/roslaunch-daiuske-VirtualBox-2157.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daiuske-VirtualBox:33287/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /usb_cam/autofocus: True
 * /usb_cam/brightness: 32
 * /usb_cam/contrast: 32
 * /usb_cam/focus: 51
 * /usb_cam/framerate: 30
 * /usb_cam/image_height: 960
 * /usb_cam/image_width: 1280
 * /usb_cam/pixel_format: mjpeg
 * /usb_cam/saturation: 32
 * /usb_cam/video_device: /dev/video0

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

auto-starting new master
process[master]: started with pid [2199]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 23a6245e-96e6-11e9-8d4a-080027134c6c
process[rosout-1]: started with pid [2212]
started core service [/rosout]
process[usb_cam-2]: started with pid [2215]
process[image_view-3]: started with pid [2216]
[ INFO] [1561425082.047664101]: using default calibration URL
[ INFO] [1561425082.047859849]: camera calibration URL: file:///home/daiuske/.ros/camera_info/head_camera.yaml
[ INFO] [1561425082.047982557]: Unable to open camera calibration file [/home/daiuske/.ros/camera_info/head_camera.yaml]
[ WARN] [1561425082.048060375]: Camera calibration file /home/daiuske/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1561425082.048184743]: Starting 'head_camera' (/dev/video0) at 1280x960 via mmap (mjpeg) at 30 FPS
[ERROR] [1561425082.322147180]: VIDIOC_STREAMON error 5, Input/output error
[ INFO] [1561425082.362763473]: Using transport "raw"
[usb_cam-2] process has died [pid 2215, exit code 1, cmd /opt/ros/kinetic/lib/usb_cam/usb_cam_node /usb_cam/image_raw:=/usb_cam/image_raw __name:=usb_cam __log:=/home/daiuske/.ros/log/23a6245e-96e6-11e9-8d4a-080027134c6c/usb_cam-2.log].
log file: /home/daiuske/.ros/log/23a6245e-96e6-11e9-8d4a-080027134c6c/usb_cam-2*.log

RyodoTanaka, 2019-06-25:
@daisuke0203 
メールで届いていたものの，こちらの投稿が消えてしまっていたので，
メールよりコピーして引用します．

下記，引用

迅速な対応ありがとうございます。
以下、$ roslaunch chapter5 usb_cam.launchを実行した結果です。
daiuske@daiuske-VirtualBox:~$ roslaunch chapter5 usb_cam.launch
... logging to /home/daiuske/.ros/log/23a6245e-96e6-11e9-8d4a-080027134c6c/roslaunch-daiuske-VirtualBox-2157.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daiuske-VirtualBox:33287/
SUMMARY

PARAMETERS

    /image_view/autosize: True
    /rosdistro: kinetic
    /rosversion: 1.12.14
    /usb_cam/autofocus: True
    /usb_cam/brightness: 32
    /usb_cam/contrast: 32
    /usb_cam/focus: 51
    /usb_cam/framerate: 30
    /usb_cam/image_height: 960
    /usb_cam/image_width: 1280
    /usb_cam/pixel_format: mjpeg
    /usb_cam/saturation: 32
    /usb_cam/video_device: /dev/video0

NODES
/
image_view (image_view/image_view)
usb_cam (usb_cam/usb_cam_node)

auto-starting new master
process[master]: started with pid [2199]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 23a6245e-96e6-11e9-8d4a-080027134c6c
process[rosout-1]: started with pid [2212]
started core service [/rosout]
process[usb_cam-2]: started with pid [2215]
process[image_view-3]: started with pid [2216]
[ INFO] [1561425082.047664101]: using default calibration URL
[ INFO] [1561425082.047859849]: camera calibration URL: file:///home/daiuske/.ros/camera_info/head_camera.yaml
[ INFO] [1561425082.047982557]: Unable to open camera calibration file [/home/daiuske/.ros/camera_info/head_camera.yaml]
[ WARN] [1561425082.048060375]: Camera calibration file /home/daiuske/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1561425082.048184743]: Starting 'head_camera' (/dev/video0) at 1280x960 via mmap (mjpeg) at 30 FPS
[ERROR] [1561425082.322147180]: VIDIOC_STREAMON error 5, Input/output error
[ INFO] [1561425082.362763473]: Using transport "raw"
[usb_cam-2] process has died [pid 2215, exit code 1, cmd /opt/ros/kinetic/lib/usb_cam/usb_cam_node /usb_cam/image_raw:=/usb_cam/image_raw __name:=usb_cam __log:=/home/daiuske/.ros/log/23a6245e-96e6-11e9-8d4a-080027134c6c/usb_cam-2.log].
log file: /home/daiuske/.ros/log/23a6245e-96e6-11e9-8d4a-080027134c6c/usb_cam-2*.log

RyodoTanaka, 2019-06-25:
@daisuke0203 
手元に同じ Logicool C270 があったので，試しました．
おっしゃるとおり，同様のエラーで起動しませんでした．
下記に原因と解決策を示します．
###### 動作環境
Ubuntu 18.04 LTS
ROS melodic
Kernel : 4.18.0-22-generic

## 前提知識
これから記載する原因，解決策には，カメラ映像のフォーマット形式の指定が大きく関わります．
そこで，linux (ubuntu)におけるこれらの情報の確認の仕方について述べます．
#### v4l-utl のインストール
UbuntuでUSB接続されたUSBカメラの情報を確認するには，`v4l2-ctl`コマンドを利用します．
インストールされていない場合は
```bash
$ sudo apt install v4l-utils
```
でインストールしてください．
#### 利用できるカメラ情報の取得
インストール後，カメラ情報を知るためには
カメラがどこにマウントされているのか知る必要があります．
```bash
$ v4l2-ctl --list-devices
```
というコマンドで，下記のように利用できるカメラの情報が表示できます．
```log
UVC Camera (046d:0825) (usb-0000:00:14.0-1):
	/dev/video2
	/dev/video3

Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-12):
	/dev/video0
	/dev/video1
```
上記の例では，ノートPC内蔵カメラとUSBカメラの２つが表示されています．
今回接続している Logicool C270 は上の UVC Cameraというものなので，カメラがマウントされているのは`/dev/video2`ということになります．

#### カメラ情報の詳細の取得
カメラのマウント場所が分かったので，下記コマンドでマウントされているカメラで利用可能な
フォーマット形式，解像度，フレームレート
を確認します．
今回は前述の通り，カメラのマウントポイントが`/dev/video2`だったのでそのように指定しています．
内蔵カメラ等が無い場合は，おそらく`/dev/video0`になるかと思います．適宜読み替えてください．
```bash
$ v4l2-ctl -d /dev/video2 --list-formats-ext
```
いずれにせよ，上記コマンドで以下のような結果を得ました．
```log
octl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
[... 298 lines cut ...]
    <!-- uvc_cam -->
    <node name="uvc_camera" pkg="uvc_camera" type="uvc_camera_node" output="screen">
      <remap from="image_raw" to="/uvc_camera/image_raw"/>
      <remap from="image_raw/compressed" to="/uvc_camera/image_raw/compressed"/>
      <remap from="camera_info" to="/uvc_camera/camera_info"/>
      <param name="device" value="/dev/video2"/>
      <param name="fps" value="30"/>
      <param name="width" value="640"/>
      <param name="height" value="480"/>
      <param name="format" value="jpeg"/>
    </node>
```
完了したら，下記コマンドで実行してください．
```bash
$ roslaunch chapter5 usb_cam.launch
```
うまく起動すれば，`/uvc_camera/image_raw`や`/uvc_camera/image_raw/compressed`等のトピックがPublishされます．
確認方法は，Rvizを起動し，`Add`->`By topic`タブ->`/uvc_camera/image_raw`の下のImageの洗濯ボタンを`raw`や`compressed`に合わせてクリックすると，映像が確認できます．

## まとめ
長文，そして大幅な変更となってしまい，大変申し訳ありません．
急ぎソースコードの修正や確認作業を行います．
その他ご不明点等ございましたらいつでもお問い合わせください．
この度はご迷惑をお掛けし，申し訳ありませんでした．
何卒よろしくお願いします．

RyodoTanaka, 2019-06-25:
@MoriKen254 
大幅に修正が必要なことが判明しました．
急ぎIssueを立てますので，修正の指南等いただけると幸いです．
何卒よろしくお願いします．

daisuke0203, 2019-06-25:
詳細な解決策のご提示ありがとうございます。しかし、残念ながら
解決策２つとも試しましたが、カメラが起動しません（LEDが消えたまま）。
もう少しだけお付き合い頂けると助かります。
以下、解決策２における実行結果です（解決策１のエラーは前回とほぼ同様）。
`$ roslaunch chapter5 usb_cam.launch
... logging to /home/daiuske/.ros/log/7e2ccd52-9735-11e9-95d1-080027134c6c/roslaunch-daiuske-VirtualBox-9118.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daiuske-VirtualBox:36363/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /uvc_camera/device: /dev/video0
 * /uvc_camera/format: jpeg
 * /uvc_camera/fps: 30
 * /uvc_camera/height: 480
 * /uvc_camera/width: 640

NODES
  /
    uvc_camera (uvc_camera/uvc_camera_node)

ROS_MASTER_URI=http://localhost:11311

process[uvc_camera-1]: started with pid [9135]
[ INFO] [1561475099.154578921]: using default calibration URL
[ INFO] [1561475099.154871094]: camera calibration URL: file:///home/daiuske/.ros/camera_info/camera.yaml
[ INFO] [1561475099.156699876]: Unable to open camera calibration file [/home/daiuske/.ros/camera_info/camera.yaml]
[ WARN] [1561475099.156822884]: Camera calibration file /home/daiuske/.ros/camera_info/camera.yaml not found.
opening /dev/video0
pixfmt 0 = 'YUYV' desc = 'YUYV 4:2:2'
  discrete: 640x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 160x120:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 176x144:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x176:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/15 1/10 1/5 
  discrete: 960x720:   1/10 1/5 
  discrete: 1024x576:   1/10 1/5 
  discrete: 1184x656:   1/10 1/5 
  discrete: 1280x720:   2/15 1/5 
  discrete: 1280x960:   2/15 1/5 
pixfmt 1 = 'MJPG' desc = 'Motion-JPEG'
  discrete: 640x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 160x120:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 176x144:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x176:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1024x576:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1184x656:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x960:   1/30 1/25 1/20 1/15 1/10 1/5 
  int (Brightness, 0, id = 980900): 0 to 255 (1)
  int (Contrast, 0, id = 980901): 0 to 255 (1)
  int (Saturation, 0, id = 980902): 0 to 255 (1)
  bool (White Balance Temperature, Auto, 0, id = 98090c): 0 to 1 (1)
  int (Gain, 0, id = 980913): 0 to 255 (1)
  menu (Power Line Frequency, 0, id = 980918): 0 to 2 (1)
    0: Disabled
    1: 50 Hz
    2: 60 Hz
  int (White Balance Temperature, 16, id = 98091a): 0 to 10000 (10)
  int (Sharpness, 0, id = 98091b): 0 to 255 (1)
  int (Backlight Compensation, 0, id = 98091c): 0 to 1 (1)
  menu (Exposure, Auto, 0, id = 9a0901): 0 to 3 (1)
  int (Exposure (Absolute), 16, id = 9a0902): 1 to 10000 (1)
  bool (Exposure, Auto Priority, 0, id = 9a0903): 0 to 1 (1)
  menu (LED1 Mode, 0, id = a046d05): 0 to 3 (1)
    0: Off
    1: On
    2: Blink
    3: Auto
  int (LED1 Frequency, 0, id = a046d06): 0 to 131 (1)
terminate called after throwing an instance of 'std::runtime_error'
  what():  unable to start capture
[uvc_camera-1] process has died [pid 9135, exit code -6, cmd /opt/ros/kinetic/lib/uvc_camera/uvc_camera_node image_raw:=/uvc_camera/image_raw image_raw/compressed:=/uvc_camera/image_raw/compressed camera_info:=/uvc_camera/camera_info __name:=uvc_camera __log:=/home/daiuske/.ros/log/7e2ccd52-9735-11e9-95d1-080027134c6c/uvc_camera-1.log].
log file: /home/daiuske/.ros/log/7e2ccd52-9735-11e9-95d1-080027134c6c/uvc_camera-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
ネットでも自分と似たエラーを見つけ、次のような事を試しましたが結果は同じままです。
・再起動する
・sudo apt-get update
sudo apt-get upgrade を試す
・USBポートを変える

何か他に解決方法ありましたらよろしくお願いします。

RyodoTanaka, 2019-06-25:
@daisuke0203 
ご検討いただきありがとうございます．
また，解決に至れず申し訳ありません．

手元の環境でmelodicだけではなく，kineticでも試してみました．
が，私の環境では同様のエラーを再現できませんでした．
しかし，他の方の環境で同様のエラーが再現したため，現在原因を調査中です．
大変申し訳ありませんが，もう少々お待ちください．

解決策になるかは不明ですが，試しにパラメータとして`format`に与えている値を`rgb`に設定指定炊いた場合，どのようになりますか？

お手数とご迷惑をおかけしますが，よろしくお願いします．

RyodoTanaka, 2019-06-26:
@daisuke0203 
お使いのLogicool C270で`uvc_camera`パッケージの起動時に`format`パラメータを`rgb`に設定することで起動する確認が取れました．
ですので，`chapter5`の`usb_camera.launch`ファイルを下記のように修正し，起動を確認していただけると幸いです．
launchファイルの起動後のカメラ映像の確認方法は以前と同様，
Rvizを起動し，`Add`->`By topic`タブ->`/uvc_camera/image_raw`の下のImageの選択ボタンを`raw`や`compressed`に合わせてクリックすると，映像が確認できます．
```xml
<launch>
    <!-- uvc_cam -->
    <node name="uvc_camera" pkg="uvc_camera" type="uvc_camera_node" output="screen">
      <remap from="image_raw" to="/uvc_camera/image_raw"/>
      <remap from="image_raw/compressed" to="/uvc_camera/image_raw/compressed"/>
      <remap from="camera_info" to="/uvc_camera/camera_info"/>
      <param name="device" value="/dev/video0"/>
      <param name="fps" value="30"/>
      <param name="width" value="640"/>
      <param name="height" value="480"/>
      <param name="format" value="rgb"/>
    </node>
</launch>
```
以上，よろしくお願いします．
無事起動が確認取れましたら，こちらでソースコードの修正，本の記載事項の更新を予定しておりますので，よろしくお願いします．

daisuke0203, 2019-06-27:
上記のlaunchファイルに修正しましたが, jpegの時と同じエラーが出てしまいます

RyodoTanaka, 2019-06-27:
@daisuke0203 
なるほど...
私が再アップロードした投稿（https://github.com/Nishida-Lab/rosbook_pkgs/issues/52#issuecomment-505274838）を拝見していた所，VMWareを使用されているかと思います．
これが直接の原因であるということは断言できませんが，ROSのパッケージの問題なのか，それ以外が原因なのかを切り分けさせていただきたいです．
そこでお手数なのですが，Ubuntuに標準でインストールされている Cheese というアプリを使ってUSBカメラが起動できるか確認していただきたいです．
エラーログを拝見する限りおそらく起動はすると思うのですが，ご協力いただけると幸いです．
お手数をお掛けしますが，よろしくお願いします．

Cheese の使い方に関しては書きURLが参考になるかと思います．
よろしくお願いします．
https://gihyo.jp/admin/serial/01/ubuntu-recipe/0147

daisuke0203, 2019-06-27:
Cheeseでも以下のようなエラーが出て起動しません.
`daiuske@daiuske-VirtualBox:~$ cheese
libv4l2: error turning on stream: 入力/出力エラーです

(cheese:7073): cheese-WARNING **: リソースから読み込むことができません: gstv4l2bufferpool.c(1054): gst_v4l2_buffer_pool_poll (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1:
poll error 1: 入力/出力エラーです (5)

libv4l2: error turning on stream: 入力/出力エラーです`

Windows10では普通に起動できたのでカメラの問題では無いようです.

RyodoTanaka, 2019-06-27:
@daisuke0203 
検証ありがとうございます！
となると，今回の問題はROSのカメラドライバの前に
VirtualBoxを使用していることに起因する問題がまずあるようですね！
とりあえず問題の切り分けができてよかったです．

VirtualBoxでカメラにアクセスできない問題について調査します．
今すぐに解決策等を提示できずすみません．
何か糸口を見つけ次第，こちらにコメントします．
お時間かかっており，申し訳ありませんが，よろしくお願いします．

RyodoTanaka, 2019-06-27:
@daisuke0203 
こちらのサイトが参考になりそうです．
よろしければお試しください．
https://yatta47.hateblo.jp/entry/2019/01/12/202843

daisuke0203, 2019-06-27:
紹介していただいたサイトに従ってやってみました.
```
daiuske@daiuske-VirtualBox:~$ v4l2-ctl --list-devices
UVC Camera (046d:0825) (usb-0000:00:06.0-2):
	/dev/video0
```
デバイスとしては認識されているみたいですが, この状態でCheeseを使ってもカメラは起動しません.
問題解決にご協力頂きとても助かります.  自分でも少しでも早く解決できるよう頑張ります. よろしくお願いします.

RyodoTanaka, 2019-06-27:
@daisuke0203 
早速の検証ありがとうございます．
大変申し訳ありません．当方，デュアルブート環境でUbuntuをインストールしているため，すぐに同じ環境を構築し，検証する事が難しい状態です．
解決になかなか至れず申し訳ありません．
進展ありましたらこちらにコメントいたしますので，何卒よろしくお願いします．

RyodoTanaka, 2019-06-27:
どなたか本件についての見識があれば，ご協力いただけると幸いです．
よろしくお願いします．

RyodoTanaka, 2019-06-27:
@daisuke0203 
お手数ですが
- VirtualBoxのバージョン
- Windowsのバージョン
をお伺いしてもよろしいですか？
よろしくお願いします．

現在Twitterにて識者の方の意見をいただこうとしているところです．
https://twitter.com/RyodoTanaka/status/1144160823004106753

よろしくお願いします．

RyodoTanaka, 2019-06-27:
@daisuke0203 
Twitterにていくつか良いサイトを教えていただいたので共有します．
手元で試したわけではないので申し訳ないですが，ご参考になれば幸いです．

一通りのことが書かれてありそうなQiitaの記事
https://qiita.com/civic/items/684c4b82428feb0c4ae1

かなり詳細な設定情報まで書かれている記事
https://unskilled.site/virtualbox-64bit-linux/

daisuke0203, 2019-06-27:
Qiitaの記事を参考にしたところ,
・Cheese
・`$ roslaunch chapter5 usb_cam.launch`
・`$ rosrun usb_cam usb_cam node`
の全てにおいてカメラが正常に起動し問題を解決できました!
VirtualBoxのUSBの設定がデフォルトでUSB1.1になっていました. これをUSB3.0に変更すると全て解決出来ました. 

自分の無知のためにここまで議論を引き延ばしてしまい本当に申し訳ないです. 皆さんのお力がなければ解決はもっと先になっていました. そして, @RyodoTanaka 氏には約5日間, 最初から最後まで神対応していただきました. 本当に感謝しています. ありがとうございました.

RyodoTanaka, 2019-06-27:
@daisuke0203 
おおお！
解決されたのですね！
本当に良かったです！

本Issueでは、私どもも確認できていなかったミスを確認させていただくことができ、大変助かりました。
また、普段触っていないVBの技術習得にもなったので良かったです。
長い間お付き合い頂き、本当にすみませんありがとうございました。
何よりIssue建て本当にありがとうございました。

MoriKen254, 2019-06-29:
@daisuke0203 さん
この度は issue でのご報告ありがとうございます。

今後の他の読者で同様の問題が起きたときにスムーズに解決できるよう、今回の issue での対策についてノウハウ蓄積の歯止めをかけたく存じております。

そこで、最終的に起動が成功した launch ファイルの内容をここに共有いただけないでしょうか？そのファイルは、もとのサンプルファイルと全く同様のものであればVBの設定のみが問題であったと思われますので、注釈として記載します。

一方で、画像フォーマット等の設定を変更した上で、かつVBの設定も修正したということであれば、我々の書籍の内容のエラッタを発行したいと思います。

大変ご多忙のところ申し訳ございませんが、最後のひと押しだけ、今後のROSデベロッパの皆様のため、お付き合いいただけましたら幸いです。

宜しくおねがいいたします。

daisuke0203, 2019-06-29:
もとのサンプルファイルでも問題なく起動できたのでVBの設定のみの問題だったと思います.
大変ご迷惑をおかけしてすみませんでした. よろしくお願いします.

MoriKen254, 2019-06-29:
承知しました。ありがとうございます。参考にさせていただきますね。

改めまして、本件クローズといたします。

closed 2019-06-29 by MoriKen254