Nishida-Lab/rosbook_pkgs #53: Chapter5. USBカメラの起動の問題と修正

RyodoTanaka, 2019-06-25:
#52 で定義されている問題です．
USBカメラが`usb_cam`パッケージだと`MPEG`形式で起動できない等の問題があります．
ソースコード，本の記載に大幅な修正が必要と考えられます．
詳細は #52 に記載していますが，大雑把に述べると下記のとおりです．

### usb_camパッケージでは起動しないフォーマット形式がある
USBカメラのうち多くが利用しているフォーマット形式である`MPEG`形式でlaunchしようとした場合，起動に失敗するケースがあるようです．
また，この問題についてはかなり前からROS Wikiでも討論されている内容でした.....orz
https://github.com/ros-drivers/usb_cam/issues/38
http://answers.ros.org/question/217382/garbled-image-problem-on-usb_cam/

解決策として，本とソースコードともに`MPEG`形式を使用しないか，
`MPEG`形式でもlaunch可能な`uvc_camera`パッケージの利用に変更する必要があると考えます．

### image_viewがうまく起動しない
こちらについては，chapter5に記載されているlaunchファイルをそのまま使用しましたが，`image_raw/raw`や`image_raw/compressed`がPublishされないばかりか，Windowも起動せず，カメラの映像が確認できない状態になっていました．
本Chapterでの目的はカメラ映像の目視が主かと考えますので，本記述を削除し，RqtもしくはRvizで映像を確認するよう修正すべきと考えます．

## まとめ
私が現状で確認している問題点と修正案は以上です．
本に記載されている内容から大きく変更が必要な点，記載されているlaunchファイルではあまりにも起動しない部分が多い点があるため，執筆頂いた方に確認をお願いしたく思います．
お手数とご迷惑をお掛けしますが，よろしくお願いします．

MoriKen254, 2019-06-25:
@RyodoTanaka 
ありがとうございます。迅速なご対応、とても助かります。

論点は理解しました。

### 注意点

以降で検討する変更点について、以下の点について確認したいです。ご意見願います。

- Logicool C270 以外でも同一のフローで正常動作する
- 本パッケージ以外への影響がないと期待できる
- CI が通る

### 対策
まずは、できるだけ追加の処置が少なくなる方法で行きましょう。
つまり、大方針としては下記の通り。

- yuyv フォーマットを利用する。
- RViz で表示する。（画像を表示できる RViz ファイルをこちらでchapter5内にて提供する）
- テキストはまずは本リポジトリ上のエラッタにて対応。次回改版 or 増刷時に変更可能かを出版社に打診。
- 本件は動作不良ですし、西田先生もここ確認していると思うので、Noのお達しが出ない限りは、直ちに実施して良いと思います。
- MPEG フォーマットでの実行については、書き始めるとボリュームが増すので、とりあえず本リポジトリ内でのエラッタで補足にしましょう。本採用を検討するのは、次回改版時が妥当と考えます。

### 処置
色々やらなければならないことが増えたのは理解しました。以下の要素に分解できませんか？

- 自動化が可能な要素
  - rosdep 等で解決可能な要素 (v4l-utils とか、うまく自動化できないか？)
- 読者が実際にコマンドを打って確認しなければならないところ。

上記が整理できれば、前者は設定ファイルで回避し、後者はエラッタの本文で回避できます。

### お願い
@RyodoTanaka 君、ここまで検証したもらったということなら、動作するソースコードや設定ファイルをお持ちであると推察します。

つきまして、まずは上記で議論した方針で正常動作するコードのPRを出してもらえますか？エラッタの本文が辛そうなら、私が担当してもOKです。

@TakeshiNishida 先生、Logicool C270 を購入していただけませんか？2,000円程度で、画像認識技術の教育用途にも使用できると思います。再現性確認のため、研究室の学生さんにも、動作確認してもらえたらとも考えております。

AtsukiYokota, 2019-06-25:
Logicool C270ですが，研究室で同じ型のものが見つかりました．

簡単に私の環境で簡単な動作確認だけしたので取り急ぎ共有します．

### 環境
- Ubuntu 16.04
- ROS Kinetic

### Cheeseによる動作確認
C270とPCをUSB接続をした後に，Cheeseというソフトウェアを使ってカメラ映像を取得することは可能でした．

### usb_camによる動作確認
すでに議論に上がっていた通り，mjpegではVIDIOC_S_FMT error 5, Input/output errorが出ていたため，次のlaunchを用いてyuyvで試したところ，warningは出ているものの画像が/usb_cam/image_rawにpublishされていることを確認しました．
```xml:usb_cam.launch
<launch>
    <!--  usb_camノードを実行する   -->
    <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen">
          <remap from="/usb_cam/image_raw" to="/usb_cam/image_raw"/>
          <param name="video_device" value="/dev/video0" />
          <param name="image_width" value="640" />
          <param name="image_height" value="480" />
          <param name="pixel_format" value="yuyv" />
          <param name="framerate" value="30" />
          <param name="contrast" value="32" />
          <param name="brightness" value="32" />
          <param name="saturation" value="32" />
          <param name="autofocus" value="true" />
          <param name="focus" value="51" />
    </node>
</launch>
```
ログは次の通りです．
```
yokota@yokota-desktop: ~/Dropbox/works/rosbook_ws$ roslaunch chapter5 usb_cam.launch 
... logging to /home/yokota/.ros/log/ad7fdfec-9760-11e9-a28c-09b1db5f4a91/roslaunch-yokota-desktop-29785.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yokota-desktop:34911/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /usb_cam/autofocus: True
 * /usb_cam/brightness: 32
 * /usb_cam/contrast: 32
 * /usb_cam/focus: 51
 * /usb_cam/framerate: 30
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/saturation: 32
 * /usb_cam/video_device: /dev/video0

NODES
  /
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

process[usb_cam-1]: started with pid [29802]
[ INFO] [1561477719.974547914]: using default calibration URL
[ INFO] [1561477719.974595476]: camera calibration URL: file:///home/yokota/.ros/camera_info/head_camera.yaml
[ INFO] [1561477719.974660160]: Unable to open camera calibration file [/home/yokota/.ros/camera_info/head_camera.yaml]
[ WARN] [1561477719.974687139]: Camera calibration file /home/yokota/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1561477719.974714655]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS
[... 92 lines cut ...]
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1024x576:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1184x656:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x960:   1/30 1/25 1/20 1/15 1/10 1/5 
terminate called after throwing an instance of 'std::runtime_error'
  what():  couldn't set format
[uvc_camera-1] process has died [pid 31061, exit code -6, cmd /opt/ros/kinetic/lib/uvc_camera/uvc_camera_node image_raw:=/uvc_camera/image_raw image_raw/compressed:=/uvc_camera/image_raw/compressed camera_info:=/uvc_camera/camera_info __name:=uvc_camera __log:=/home/yokota/.ros/log/ad7fdfec-9760-11e9-a28c-09b1db5f4a91/uvc_camera-1.log].
log file: /home/yokota/.ros/log/ad7fdfec-9760-11e9-a28c-09b1db5f4a91/uvc_camera-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
簡単な確認のみで申し訳ありませんが取り急ぎ，よろしくお願いします．

RyodoTanaka, 2019-06-25:
@AtsukiYokota 
検証ありがとうございます．
#52 で質問者の @daisuke0203 さんがおっしゃっているエラーと同様のエラーと考えます．．．
私の環境と異なる点は，melodic か kinetic という点です．
そこで，手元にあるkinetic環境で同様の手続きを行ったのですが，エラーが再現しませんでした...
なので今ここで決定的な解決策を申し上げられないのですが，試しに`format`を`rgb`に変更した場合どうなるのか検証頂いても良いですか？
よろしくお願いします．

RyodoTanaka, 2019-06-25:
@MoriKen254 
後出しになってしまったのですが， @AtsukiYokota さんのご協力のおかげでエラーを確認できました．
そのため，原因の究明と解決策をみつけるまでブランチへのアップロードを少し伸ばそうと思います．
また，本問題が解決次第，エラッタについては検討します．
すみませんが，よろしくお願いします．

RyodoTanaka, 2019-06-25:
@MoriKen254 
また，`uvc_camera`パッケージについては
READMEに現在は新しいパッケージがあるとの記載があるため，再考する必要がありそうです．
調査に少し時間が掛かりそうです．（長くて一週間）
なので，とりあえずは解決策を見つけた上で，修正すべきlaunchファイルの内容を確定し，ブランチを切ってアップロードした後にエラッタについて検証します．
よろしくお願いします．

MoriKen254, 2019-06-25:
諸々ありがとうございます。理解しました。

http://wiki.ros.org/libuvc_camera があるのですね。

後継版があるなら、こちらに移行すべきですね。お手すきの際に検証してもらえると助かります。納得の行く手段で修正できればと思います。

AtsukiYokota, 2019-06-26:
@RyodoTanaka さん
formatをrgbに変更したところ画像が正しくpublishされることを確認しました．
```xml
<launch>
    <!-- uvc_cam -->
    <node name="uvc_camera" pkg="uvc_camera" type="uvc_camera_node" output="screen">
      <remap from="image_raw" to="/uvc_camera/image_raw"/>
      <remap from="image_raw/compressed" to="/uvc_camera/image_raw/compressed"/>
      <remap from="camera_info" to="/uvc_camera/camera_info"/>
      <param name="device" value="/dev/video0"/>
      <param name="fps" value="30"/>
      <param name="width" value="640"/>
      <param name="height" value="480"/>
      <param name="format" value="rgb"/>
    </node>

    <!--  image_viewノードを実行する   -->
    <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
      <remap from="image" to="/uvc_camera/image_raw"/>
    <param name="autosize" value="true"/>
    </node>
</launch>
```
image_viewはあってもなくても動作しました．ログは次の通りです．
```
yokota@yokota-desktop: ~/Dropbox/works/rosbook_ws$ roslaunch chapter5 usb_cam.launch 
... logging to /home/yokota/.ros/log/05c8174c-97bf-11e9-b15a-7085c25e6ccf/roslaunch-yokota-desktop-23855.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yokota-desktop:35181/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /uvc_camera/device: /dev/video0
 * /uvc_camera/format: rgb
 * /uvc_camera/fps: 30
 * /uvc_camera/height: 480
 * /uvc_camera/width: 640

NODES
  /
    image_view (image_view/image_view)
    uvc_camera (uvc_camera/uvc_camera_node)

ROS_MASTER_URI=http://localhost:11311

process[uvc_camera-1]: started with pid [23872]
process[image_view-2]: started with pid [23873]
[ INFO] [1561518364.080702819]: using default calibration URL
[ INFO] [1561518364.080808018]: camera calibration URL: file:///home/yokota/.ros/camera_info/camera.yaml
[ INFO] [1561518364.080897509]: Unable to open camera calibration file [/home/yokota/.ros/camera_info/camera.yaml]
[ WARN] [1561518364.080930596]: Camera calibration file /home/yokota/.ros/camera_info/camera.yaml not found.
opening /dev/video0

(image_view:23873): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(image_view:23873): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",

(image_view:23873): Gtk-WARNING **: Unable to locate theme engine in module_path: "pixmap",
[ INFO] [1561518364.214901255]: Using transport "raw"
pixfmt 0 = 'YUYV' desc = 'YUYV 4:2:2'
  discrete: 640x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 160x120:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 176x144:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x176:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/25 1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/15 1/10 1/5 
  discrete: 960x720:   1/10 1/5 
  discrete: 1024x576:   1/10 1/5 
  discrete: 1184x656:   1/10 1/5 
  discrete: 1280x720:   1/10 1/5 
  discrete: 1280x960:   2/15 1/5 
pixfmt 1 = 'MJPG' desc = 'Motion-JPEG'
  discrete: 640x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 160x120:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 176x144:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x176:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 320x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 352x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 432x240:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 544x288:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 640x360:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 752x416:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x448:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 800x600:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 864x480:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x544:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 960x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1024x576:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1184x656:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x720:   1/30 1/25 1/20 1/15 1/10 1/5 
  discrete: 1280x960:   1/30 1/25 1/20 1/15 1/10 1/5 
  int (Brightness, 0, id = 980900): 0 to 255 (1)
  int (Contrast, 0, id = 980901): 0 to 255 (1)
  int (Saturation, 0, id = 980902): 0 to 255 (1)
  bool (White Balance Temperature, Auto, 0, id = 98090c): 0 to 1 (1)
  int (Gain, 0, id = 980913): 0 to 255 (1)
  menu (Power Line Frequency, 0, id = 980918): 0 to 3 (1)
    0: Disabled
    1: 50 Hz
    2: 60 Hz
    3: Auto
  int (White Balance Temperature, 16, id = 98091a): 0 to 10000 (10)
  int (Sharpness, 0, id = 98091b): 0 to 255 (1)
  int (Backlight Compensation, 0, id = 98091c): 0 to 1 (1)
  menu (Exposure, Auto, 0, id = 9a0901): 0 to 3 (1)
  int (Exposure (Absolute), 16, id = 9a0902): 1 to 10000 (1)
  bool (Exposure, Auto Priority, 0, id = 9a0903): 0 to 1 (1)
select timeout in grab
[ WARN] [1561518368.262451039]: Left-clicking no longer saves images. Right-click instead.
```

RyodoTanaka, 2019-06-26:
@MoriKen254 
承知しました．
検証してまたアップします．

RyodoTanaka, 2019-06-26:
@AtsukiYokota 
検証ありがとうございます！！！！
とても助かります．
とりあえず #52 で質問をくださった @daisuke0203 さんには，`format`を`rgb`に設定する方向でお願いしようと思います．

RyodoTanaka, 2019-06-26:
@AtsukiYokota 
今確認していて気づいたのですが，載せていただいているlaunchファイルには`image_view`パッケージの起動も記載されているようです．
こちらの起動についてはいかがでしたか？

AtsukiYokota, 2019-06-26:
@RyodoTanaka さん
載せたlaunchを起動すると`image_view`も正しく起動し`RViz`同様に画像が表示されました．

RyodoTanaka, 2019-06-26:
@AtsukiYokota 
なんと！
逆に僕の環境では動かなかったんですよね...

AtsukiYokota, 2019-06-26:
@RyodoTanaka さん
おや，，？こちらでは以下の画像のような感じで動いてます．
ログが見切れていますが，https://github.com/Nishida-Lab/rosbook_pkgs/issues/53#issuecomment-505699425 のログと同じだと思います．
このlaunchとログをそちらの環境のものと見比べて何か違う点はありますか？

[image]

RyodoTanaka, 2019-06-26:
@AtsukiYokota 
ちょっと今ログを取れないんですが，エラーメッセージと真っ黒な画面が出て終わりなんですよね．
ただ，この表示画面を見るに，OpenCVのインストールに関係してそうですね．
OpenCVのインストールに関しては，確かに試した環境には入れてなかったかもしれないです．
確認してみます．
ありがとうございます！

AtsukiYokota, 2019-06-26:
@RyodoTanaka さん
ありがとうございます，ご確認よろしくお願いします．
一応私の環境について補足だけしておきますね．
- `sudo apt install ros-kinetic-desktop-full`をした時に一緒に入ってきたもの以外のOpenCVは別途手動でインストール等をしていません
- `uvc_camera`はインストールしていなかったので新たに`sudo apt install ros-kinetic-uvc-camera`で最新のものをインストールしました

RyodoTanaka, 2019-06-26:
@AtsukiYokota 
なるほどとても助かります．
ありがとうございます．

MoriKen254, 2019-06-26:
みなさん、お疲れ様です。uvc_camera で動けば、とりあえずそれで行きましょうね。

MoriKen254, 2019-06-27:
結論、https://github.com/Nishida-Lab/rosbook_pkgs/issues/52 より、VB に依存する問題でした、でよかったでしょうか。

実機でも紛糾していたところを見ると、コードへの変更もある程度必要ですか？

いずれにせよ、現状の本文では少し説明が足りないかもしれないので、エラッタは出したいなと思っています。

PR等あれば、宜しくおねがいします。

MoriKen254, 2019-06-29:
VB のみの問題ということで、エラッタには注釈として追記しようと思います。何か不都合等ございましたら、連絡願います。

@RyodoTanaka くん、最大の功労者ですね。ありがとうございます。お疲れ様でした。 @AtsukiYokota   くんも、検証ありがとうございました。助かりました。

歯止めはお任せください。

RyodoTanaka, 2019-07-03:
@MoriKen254 
お返事が遅くなり，大変申し訳ありません．
#52 で解決の問題ですが，私の認識は以下のとおりでした．
### 1. Virtual Box の設定問題
Virtual Box の設定にUSB 3.0 を選択しないとカメラに接続できないという問題．
設定を変更すればうまく行くようです．

### 2. usb_cam か uvc_camera か
Virtual Box の問題に行き着く前に，私の環境と @AtsukiYokota さんの環境で，本に記載の `usb_cam`パッケージを使用し，MJPEG圧縮手法でカメラ画像を取得しようとするとエラーが起きておりました．
対処法として，
1. usb_cam のフォーマットを'yuyv`（この設定値はカメラ依存）に変更する
2. usb_camではなく，uvc_cameraパッケージを利用する．この場合，MJPEG圧縮形式ならフォーマットを`jpeg`に，それ以外は`rgb`に設定すればどちらでも起動する

となっていました．

以上のように問題点は2つあり， #52 の質問者の方はVBの問題を解決すれば，本に記載の通りで起動したとのことでしたが，私や @AtsukiYokota さんの環境では，２つめの問題が依然として残っている状態と考えています...

RyodoTanaka, 2019-07-03:
@MoriKen254 
エラッタの注釈として書いておくのが妥当なのか，なんとも言えない気がするのですが，ご意見いただけると幸いです．
よろしくお願いします．

MoriKen254, 2019-07-03:
ありがとう。そうですね、その辺も一応書いておこうと思います。

あと、私の環境でも試してみたいですね。それでも起こるようなら、もう次回増刷時に内容変えたほうがいいレベルですね。

rviz と image_view の謎現象もありましたね。uvc_camera で全部解決するなら、できれば移行したいですね。

手元に例のカメラがないので、今度研究室のを借りて検証してみます。

RyodoTanaka, 2019-07-04:
@MoriKen254 
コメントありがとうございます．
確かに，もう少しいくつかのカメラと環境で試したほうが良さそうですね．
お忙しいところありがとうございます．
よろしくお願いします．

MoriKen254, 2019-11-02:
これ、ハード依存であり、ソフト解析の書籍でのサポートは限界があるので、閉じましょうね。

closed 2019-11-02 by MoriKen254