Nishida-Lab/rosbook_pkgs #58: dynamixelサーボの認識がされない

TsuruharaKota, 2019-11-21:
dynamixelサーボをchapter5を実行して動かそうとしたのですが
$roslaunch chapter5 pan_tilt_controller.launch
を実行すると以下のようなエラーが表示されます
process[dynamixel_manager-1]: started with pid [9777]
process[dynamixel_controller_spawner_AX12-2]: started with pid [9778]
[INFO] [1574290084.330697]: pan_tilt_port controller_spawner: waiting for controller_manager dynamixel_manager to startup in global namespace...
[INFO] [1574290084.600601]: pan_tilt_port: Pinging motor IDs 2 through 2...
[ERROR] [1574290084.618914]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.639490]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.655003]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.672552]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.688977]: Exception thrown while getting attributes for motor 2 - 1020
[INFO] [1574290084.690657]: pan_tilt_port: Found 0 motors , initialization complete.
[INFO] [1574290084.706795]: dxl_tty1: Pinging motor IDs 1 through 2...
[ERROR] [1574290084.807841]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.825282]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.844558]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.862981]: Exception thrown while getting attributes for motor 2 - 1020
[ERROR] [1574290084.880637]: Exception thrown while getting attributes for motor 2 - 1020
[INFO] [1574290084.882483]: dxl_tty1: Found 0 motors , initialization complete.
[INFO] [1574290084.944417]: pan_tilt_port controller_spawner: All services are up, spawning controllers...
[WARN] [1574290084.984597]: The specified motor id is not connected and responding.
[WARN] [1574290084.985059]: Available ids: []
[WARN] [1574290084.985378]: Specified id: 2
[ERROR] [1574290084.985891]: Initialization failed. Unable to start controller pan_joint
chapter5のbaud_rateを57600に変更してtilt_jointはコメントアウトしpan_jointだけにして、xm430-w350-rを一台使用しています。
お忙しいとは思いますがよろしくお願いします

RyodoTanaka, 2019-11-21:
@TsuruharaKota 
ご質問頂き，ありがとうございます．
現状全く同じ機種を手元に持っていないため，すぐに検証することができないのですが，
他の機種のDynamixelが手元にあるため，時間が取れ次第こちらで検証させていただきます．

ただ，こちらで検証を行う前に現状でお伝えできることとして，下記があります．

### 1. モーターの検出確認
モーターの検出確認には`dynamixel_workbench_controllers`パッケージを`apt`でインストール頂き，
```
rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0
```
のようにして検出可能かチェックしていただくことができます．（`/dev/ttyUSB0`は実際に接続されているデバイス名に適宜読み替えてください．）
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

ここまでの段階で何も検出されない場合，念の為
- 改めて物理的な接続状態を確認
- `/dev/ttyUSB0`の権限の変更（`chmod`コマンドで行えます）

を行って頂けると幸いです．

### 2. 本に記載の情報との齟齬
今回は「モーターが検出されない」という問題なのでどこまで関係するかわかりませんが，
Dynamixelについては本に記載の情報から更新がかかっており， #50 でもこれに関するIssueを頂いています．もしよろしければ #50 に記載のリンク
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#ros-tutorials
もご覧ください． 

以上，お手数をお掛けしますが，よろしくお願いします．

TsuruharaKota, 2019-11-21:
find dynamixel
の実行ではdynamixelが認識されます。
また、dynamixel wizardでも認識は出来ています 。/dev/ttyUSB0への権限を与えましたが,launchを実行すると同じエラーが出ました

RyodoTanaka, 2019-11-21:
@TsuruharaKota 
検証頂き，ありがとうございます．

### 1. 問題点と原因
今，下記のROS wikiからGitHubページに飛び
http://wiki.ros.org/dynamixel_controllers
Issueを確認した所，同様のエラーを見つけました．
https://github.com/arebgun/dynamixel_motor/issues/86#

読み進めていくと， @TsuruharaKota さんと同様に Dynamixel XM430-W350-R の検出がされないとのことでした．
これに対する `dynamixel_controller` パッケージの作者の回答は

「そもそも XM430-W350-Rはドライバの対応モーターに想定していないので動かない．」

ということでした．
https://github.com/arebgun/dynamixel_motor/issues/86#issuecomment-524973678

chapter5でご紹介しているモーターは Dynamixel AX-12A であるため，これを用いる場合には問題は起こりません（動作検証済み）が，今回は上記理由により動かなかったと考えられます．

### 2. 解決方法
こちらの回答は動作を保証するものではありませんが，Dynamixel より公式のドライバーパッケージが`dynamixel_workbench*`として更新されています．
英語のマニュアルになるため，申し訳ありませんが， 5.1.2 Controllers の章からが参考になるかと思います．
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/

お手数をお掛けしますが，よろしくお願いします．

TsuruharaKota, 2019-11-21:
分かりました。dynamixel_workbenchを使用してみたいと思います。
ありがとうございました。

closed 2019-11-21 by RyodoTanaka